360Controller/360Controller #45: Lights around XBox button never stop spinning, even when connected and playing.

banzai777, 2015-01-03:
The controller connects fine with the transceiver and the XBox360 Controllers control panel see the input just fine. My controller, though, will not stop spinning and stay on the player 1 light.

Does this happen to anyone else?

I'm using OS X Yosemite 10.10.1 and version 0.14 of the XBox360 Controller driver. It stopped working altoegther after I updated to Yosemite but 0.14 fixed it. The spinning lights had been a problem before I updated to Yosemite, though.

d235j, 2015-01-03:
The lights will spin as long as the preference pane is open. This bug is a duplicate of https://github.com/d235j/360Controller/issues/35.

closed 2015-01-03 by d235j

brianhanifin, 2015-01-03:
Fyi, I have found the lights spin IF the adapter is plugged into my MacBook Pro before the controller is turned in. If I turn the controller on first the player one light operates as expected.