360Controller/360Controller #46: Controller order

Duros, 2015-01-03:
I've noticed that the controller order get messed up when a controller disconnects.

For example, there are 4 360 controllers used. Controller 3 disconnects, and you reconnect it straight after it disconnects. It will show up again as controller 3 on the indicator lights, but the game will see it as controller 4 and not as 3, while the real controller 4 will not be able to control anything form that point.

So the indicator lights on the controller work well after reconnecting, but the internal controller order gets messed up. I've tested this with 2 games so farm one of then Towerfall which works native with 360 controllers.

zaius, 2015-01-03:
I'm seeing this on towerfall and all source engine games, but disconnect and reconnects work fine on castle crashers, so I'm thinking it's maybe a game issue. Does anyone have access to towerfall on windows to test if the same thing happens there?

Duros, 2015-01-03:
I have played it on windows some weeks ago. We didn't have the same problem from what I recall. The game shows a window as soon as controller disconnects. With the native MS firmware for the 360 controller it works fine I guess. I can probably test this on a windows machine next week.

MaddTheSane, 2015-01-03:
@Duros That is… an interesting bug.
What specific games are showing this behavior? Does the game gracefully handle hot-plugging and hot-unplugging of controllers? Do you have other (non-360) controllers you can test this with?

zaius, 2015-01-03:
After playing around for a while, I think this is a towerfall bug. The player controllers don't always map correctly to the inputs. e.g. player one maps to the controller with the 3rd light. So I think something about their controller discovery is messed up on mac.

A work around is to turn off gamepad discovery, then turn off all controllers at the same time and then back on. Sometimes you have to switch controllers to get the right player back, but at least you don't lose control of any of the players.

Duros, 2015-01-04:
@MaddTheSane So far I had this with Towerfall and Megabyte Punch. I'll test Broforce and Castle Crashers tomorrow. As far as I know hot-plugging works within Towerfall, but will test at as well.
Don't have any non 360 controller.
@zaius Alright, that could be a thing. But it will be suspicious if all multiplayer games I own have the same issues. We now just turn every controller off and reconnect them in the right order. But it's still an annoying interruption. As specially when a controller has two random disconnects.

zaius, 2015-01-04:
The reason I think it's probably a game specific problem is that it works fine for me in castle crashers. It's the one game I have that works flawlessly though. I'll see if I can get my hands on a different controller and try it out on towerfall. That way we can narrow it down to a game vs controller problem.

Another potential solution would be to have some way to re-initialize all controllers in the correct order on the system prefs panel. Then you could hook that up via a script - e.g. hold the X on all controllers at the same time.

Duros, 2015-01-04:
I have tested a couple games:
Awesomenauts - X Once the controller disconnects your able to reconnect and have to restart te game.
Castle Crashers - √
The Expendabros - √
FORCED - √ Order get messed up
Megabyte Punch - √ Order get messed up
Towerfall - X Order get messed up and some characters are no longer playable.

I think like @zaius claims, that it is more game related than the firmware. Some games do pick up the controller as intended, but others don't. Maybe those games are not optimized for multiple controllers.

RodrigoCard, 2015-01-08:
I am pretty sure this is a game-specific issue, as it does not happen in some games (Castle Crashers handle this correctly). Multiple controller handling with hot plugging is a bit tricky to get right, I have been messing with that in a game i am developing right now.

Also, I dont think there is a workaround that will help with awesomenauts and other games that does not handle hot-plugging.

I suggest you to contact the respective games' developers and see if they can fix that.

zaius, 2015-01-10:
Ok, +1 to close this then. After thinking about it, the only good solution I could come up with was a re-init of all the controllers in the right order every time one reconnects, and that could just as easily break other games, so I think it should be left as is.

closed 2015-01-15 by Duros