360Controller/360Controller #47: On/Off button menu bar [New Feature]

Duros, 2015-01-03:
I don't if it is possible but a I like to see an on/off button in the menu bar to turn your controllers of in the menu bar as a new feature. Because now you have to open the config window every time you like to do that. I have no clue if this is a lot of work to implement, if so, don't bother.

brianhanifin, 2015-01-03:
I pop the battery door open a hair and immediately close it to turn the controller off quick and easy.

MaddTheSane, 2015-01-03:
I can see the hardest part for me would be getting (or making) an icon for the menu.

I think this is a good idea, but I'd like a way to toggle the presence of the menu item, as some Mac users don't have a lot of menu bar room.

zaius, 2015-01-03:
I currently pop the battery out as well, but I think it would be nice to replicate the actual xbox 360 setting, where holding down the middle Xbox button for 3 seconds turns the controller off.

That said, the windows driver doesn't do this, and things like steam use the xbox button to launch big picture mode, so maybe not a good solution for everyone.

Duros, 2015-01-04:
At a friends house we have two controller where the battery door is broken, so now have to wait or peel the tape off it turn it off. Also I prefer a more slick in doing it.
Maybe a combination like hold 'back' and 'A' at the same time for 3 seconds.
As for an icon, maybe 'borrow' the big picture logo from steam as an icon?