360Controller/360Controller #48: Xbox 360 Wireless Controller for Windows (New device)

pasha-gooff, 2015-01-03:
The problem is the same as in https://github.com/d235j/360Controller/issues/38

Xbox 360 Wireless Receiver for Windows:
              Product ID: 0x0719
              Vendor ID: 0x045e  (Microsoft Corporation)
              Version: 1.00
              Serial Number: 00000000
              Speed: Up to 12 Mb/sec
              Manufacturer: ©Microsoft
              Location ID: 0x1a110000 / 3
              Current Available (mA): 500
              Current Required (mA): 260

I could see in Windows under Parallels id of the controller:
USB\VID_045E&PID_02A1&IG_00

What should I add to the info.plist to work?

P.S. Sorry for my bad english

MaddTheSane, 2015-01-03:
Is Parallels running when you try to use your controller?
Have you tried un-plugging the adaptor or turning the controller off, then back on?

pasha-gooff, 2015-01-03:
Yes, I launched Parallels only to find the PID controller.
In OpenEmu and PPSSPP controller is determined, but the game of the steam can not see it (he specifically steam and LEGO Marvel heroes)

RodrigoCard, 2015-01-21:
Could you test your controller with this version?
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview2.pkg

Please restart once, if you have used the driver since the last time you boot your computer.

pasha-gooff, 2015-01-22:
Does not help

pasha-gooff, 2015-01-30:
Working on 10.10.2

closed 2015-01-30 by pasha-gooff