360Controller/360Controller #49: Logitech F710 Support

BeeOnline, 2015-01-06:
Hello,

Before I've upgraded to Yosemite, I used [this instruction](http://x360controller-osx-driver.blogspot.ru/2014/09/use-any-xbox-360-controller-on-osx.html) and your driver in 10.9 to play Steam games using Logitech F710 gamepad in X-Input mode and it worked great, thank you for that!

But with the upgraded driver and 10.10 the trick does not seem to work, could you please look into it?

PS: I (and many F710 users I guess) would also appreciate, if the driver supported this gamepad out-of-the-box, there seems to be no rocket-science in Xbox360DriverExtraPadSupport.command

MaddTheSane, 2015-01-06:
See #25. Although I don't know if the f710 is on the list.

MaddTheSane, 2015-01-06:
Looking at the command, it is.

And I have merged #25. Closing.

closed 2015-01-06 by MaddTheSane