360Controller/360Controller #54: Include Uninstaller

bnnoirjean, 2015-01-12:
I have read in another form that  we can use pkgutil --files com.mice.pkg.Xbox360controller; to see the files installed by your driver and uninstalled manually. I am wondering if maybe you could include some kind of uninstaller to make the process easier.
Thank you for your amazing work

Deph0, 2015-01-16:
I'd image something like adding a uninstall package to the Install360Controller project.
upgrade.sh (since it does most of the work for you already) would be pre-installation and a uninstall.sh with '/usr/sbin/pkgutil --forget com.mice.pkg.Xbox360controller'  for post-installation.

The Packages app itself just provides a uninstall.sh file inside a folder in the dmg file, which probably isn't the best way but works.

-just my two cents on this note

RodrigoCard, 2015-01-21:
I use this script to remove the driver, feel free to improve it:

```
#!/bin/sh
# pkgutil --pkgs | grep 360
# TODO: check if "com.mice.pkg.Xbox360controller" exists

# needs root permissions, run this with sudo
[ "$(whoami)" != "root" ] && exec sudo -- "$0" "$@"

# unload everrything related to this bundle
kextunload -b "com.mice.pkg.Xbox360controller"

# the daemon
rm -r "/Library/Application Support/MICE/360Daemon"
rm    "/Library/LaunchDaemons/com.mice.360Daemon.plist"
# preference pane
rm -r "/Library/PreferencePanes/Pref360Control.prefPane"
# kexts:
# old locations (system library folder)
rm -r "/System/Library/Extensions/360Controller.kext"
rm -r "/System/Library/Extensions/Wireless360Controller.kext"
rm -r "/System/Library/Extensions/WirelessGamingReceiver.kext"
# new locations (extensions shouldnt be installed on /System/... anymore)
rm -r "/Library/Extensions/360Controller.kext"
rm -r "/Library/Extensions/Wireless360Controller.kext"
rm -r "/Library/Extensions/WirelessGamingReceiver.kext"

# remove pkg ticket
pkgutil --forget com.mice.pkg.Xbox360controller

# show related files just so you can be sure there are no files left
pkgutil --files com.mice.pkg.Xbox360controller
```

RodrigoCard, 2015-01-22:
I was thinking, Maybe I should just improve this script a bit and include it with the PKG installer in a DMG file in the next release.

FranticRain, 2015-01-24:
Personally I would add the -f flag to rm `rm -rf` so that if one of the files was previously deleted, it doesn't ruin everything and break the script. It may also be helpful to `kextstat | grep 360`.

zaius, 2015-01-24:
You can also use applescript to get sudo permissions through a popup so this could run completely in the background - more details here:

http://stackoverflow.com/questions/1517183/is-there-any-graphical-sudo-for-mac-os-x

FranticRain, 2015-01-24:
In retrospect, you may not want to go with something as simple as grep 360. You could accidentally find com.apple.driver.AirPort.Brcm4360. But here is a little if snippet I made to suppress any sort of failure related to the driver not being loaded when you try to unload it.
```
kextstat | grep -g Xbox360Controller
if [ $? -eq 0 ]
then
	kextunload -b "com.mice.driver.Xbox360controller"
fi
```
It is important to note that the driver probably can't be unloaded with the .pkg. version of the address, as you originally had it set up.

RodrigoCard, 2015-01-31:
@FranticRain Yeah, I didnt notice I put "com.mice.pkg.Xbox360controller" instead of "com.mice.driver.Xbox360controller", that Was doing nothing silently and I thought it was working.

I was writing a proper uninstall script and doing some tests, and noticed that the upgrade.sh was also failing silently. **It always fails to unload the kexts**. Making it impossible to upgrade without restarting.

I am writing a new issue about that.

RodrigoCard, 2015-02-01:
I mean, not impossible to upgrade, but impossible to use the upgraded driver correctly before restarting in most cases...

RougeTomate, 2015-04-01:
So how do we safely uninstall this?

RodrigoCard, 2015-04-21:
included in eee7beb3be9b745d0e9e70cf6feec8dd967ab98c

closed 2015-04-21 by RodrigoCard