360Controller/360Controller #55: GameController Framework support

paiv, 2015-01-17:
Apple has their own framework to provide access to game controllers: [Game Controller Framework Reference](https://developer.apple.com/library/mac/documentation/GameController/Reference/GameController_RefColl/index.html)

I have tried it with Xbox360 Wireless and this driver, but nothing gets found:

    #import <GameController/GameController.h>

    for (GCController *controller in [GCController controllers]) {
        NSLog(@"CONTROLLER FOUND: %@", controller);
    }

I could access it via SDL2.
So I was wondering, what would be the effort to add support for GameController Framework?

MaddTheSane, 2015-01-17:
At a cursory glance, it doesn't look like there is a way to add non-mFi(Made for iPad/iPhone/iPod) controllers to the GameController framework.

dak180, 2016-02-04:
@MaddTheSane you may want to check out https://github.com/slembcke/CCController since it seems to add some preliminary support for this.

slime73, 2016-02-04:
CCController isn't a way to have non-MFi gamepads added to GameController.framework, it's just a GameController.framework-like wrapper over the HID API. So it's not especially useful, as each developer of each game would have to integrate it into their codebase.

FranticRain, 2018-06-01:
There is a note about this in the README, so I am going to close this down.

closed 2018-06-01 by FranticRain