360Controller/360Controller #57: Screensaver turns on while playing

AndreasBoehm, 2015-01-19:
Yesterday i tried to play "Broforce" with my brand new XBOX 360 Wireless Controller with your driver under OSX 10.10.1. It worked well, but the screensaver appeared while playing.

Is this game-related or input-device related?
I think it could be a combination of both, havent tried any other game.

maybeageek, 2015-01-19:
I think it might be input device related, it happened here too with Sonic All Stars Racing.

slime73, 2015-01-19:
That'd be up to the game's code. Typically a game will have a repeating timer that stops the screensaver from activating every few seconds, while the game is in the foreground. Games that use SDL can just call the handy `SDL_DisableScreenSaver` function. Recent versions of SDL even do that by default.

RodrigoCard, 2015-01-21:
Does it happen with other games?

khallmark, 2015-03-22:
This is an issue with controllers in general, I think. I had issues with it in every game I was playing with a dualshock4 also.

MaddTheSane, 2015-03-22:
This is less an issue with controllers and more an issue with games *not* telling the OS not to turn on the screen saver. Although this may be because older games are using outdated APIs which may not work on newer OSes.

khallmark, 2015-03-22:
I've noticed the same behavior running in Parallels.

Connect the controller to Mac, share the usb device to Parallels (in this case a DualShock 4), play games. Screen shuts off according to energy saver. It could also be an issue with Parallels too. 

It's definitely a common issue with gaming pads, whether it's each application or the system itself.

RodrigoCard, 2015-03-23:
unfortunately , as @MaddTheSane said, this is something that the games themselves should worry about. So, not a bug.

closed 2015-03-23 by RodrigoCard