360Controller/360Controller #58: Feature request/bug report: slider for adjusting axis sensitivity

heinzK1X, 2015-01-19:
Hi, 
I'm currently using the xbox 360 wireless controller to play some good ol' n64 games with openEmu. The problem is that the controls on the left joystick axes are way to sensitive, so on moving the sticks only a little bit Mario just RUNS instead of walking slowly.
So: 
A) I didn't understand what the sliders on the very left and at the bottom right are for. When I move them I see the red lines going back and forth, but it doesn't change the controls in any way. If these sliders are supposed to change sensitivity, then this is a bug-report.
B) The sliders have nothing to do with sensitivity. Then this is a feature-request for adding another slider, that would decrease sensitivity, so that little Mario can walk slowly in the future. 

Cheers! -H

Deph0, 2015-01-19:
Hi, The sliders are indeed supposed to be for sensitivity/deadzone.
Have you tried using another game (not OpenEmu)? I'm not sure if OpenEmu has their own make-shift driver for controllers.

blakekl, 2015-01-19:
There's a difference between sensitivity and deadzone. I think he's essentially asking for a way to modify the sensitivity, which would require specifying a curve.

Deadzone: the controller doesn't report input until a certain threshold is reached.
Sensitivity: the controller reports a different value than the actual input.

say your control outputs values in the 0-100 range (0 at rest). In the example above, say mario runs when the joystick reports >= 20% of input range (so 20+). If you have your deadzone set to 25%, mario will always run because the joystick will always respond with 0 until you have pushed the stick past the 25 point, at which point it sends 25 for it's value,

Setting a curve, you could set a deadzone at 25% and then change the curve to report 1 at 25% and increase steadily from there until 100% at full output.

heinzK1X, 2015-01-19:
Yes, @blakekl that's kind of what I was looking for. I also confirm that the deadzone feature is working, which is nice as my controller does not always go back to perfekt zero. 
I think the feature I'm looking for is a down-variation of the endpoint of an axis, with linear scaling along the way. E.g., an additional slider stearing values from 0.1 to 1, where the controller input value is simply multiplied by the value of the additional slider.

blakekl, 2015-01-19:
You don't really need a separate slider if you want it to always be linear. The driver would just have to translate the input values to the correct values based on the already specified deadzone. A curve allows you to do some other really cool/weird stuff. For instance, you could have the curve actually reverse the input, or make it so mario walks unless the stick is reporting full input. I'm not necessarily advocating for a curve, just pointing out the difference. A linear adjustment based on deadzone should be fairly simple to implement and would satisfy most use cases, I think.

heinzK1X, 2015-01-19:
Ok, I just entered the game again and I think you're right. If you'd simply attach the zero endpoint to the deadzone, it would be fine. The max endpoint can be the same and you should have linear scaling in between. Is this easy to implement?

FranticRain, 2015-02-09:
Adding a full bezier curve editor to the preference pane could get pretty complicated, but I am more concerned with transferring the data to and from the driver. You either send a bunch of data for the curve over, or build a curve from points at the kernel level, and neither of those sound smart. So I was thinking I'd take a look at specifically Mario 64 and develop some premade curves that the driver can implement simply. Ideally the list would be something like: Linear, Ease In, and Ease Out. Where Ease In would be the selection you would want.

blakekl, 2015-02-09:
@FranticRain 
That would work, but I think linear with adjustments for dead zone should be default, rather than selecting ease in. This would essentially make a controller with a high dead zone behave more like a normal controller in many other games. Mario 64 is one example, but there are several others. Metal Gear Solid uses a similar mechanic for walking and running. Walking is silent while running is noisy, which is a big deal in a stealth game. Here are some space sim shooters where smaller input is yaw control and larger values result in roll. 

My controller I use has a dead zone of about 25% of the stick movement, and it is very difficult to make Mario/Snake walk. If the driver reported 0 at he dead zone value and was linear from that point, you could still give greater degrees of input.

FranticRain, 2015-02-10:
Should have a fix out for this soon. [Check it out in action](https://github.com/d235j/360Controller/pull/75#issuecomment-73644629).

RodrigoCard, 2015-07-28:
this one is fixed in #75 , closing

closed 2015-07-28 by RodrigoCard