360Controller/360Controller #59: Latest Driver Release Preventing Display Sleep While Plugged In

therealbushwhack, 2015-01-20:
My Gamestop controller is FINALLY supported in this release, but I have another issue. 
As long as the controller is connected via USB, the display fails to go to sleep and I need to remember to unplug the controller so the display sleeps as expected. Is there a way to fix this so the display functions/sleeps as expected? Unplugging the controller when done with it does fix it, but where it is now attached is not the most convenient place to reach to unplug when done...

GRPyles, 2015-07-04:
I came here to confirm that I too am having this issue with it preventing sleep, but unplugging the controller does allow it to sleep. Just throwing this in to see if there is any updates planned.

RodrigoCard, 2015-07-10:
could someone else confirm this?
I will test when I have access to my xbox controllers this weekend.

GRPyles, 2015-08-06:
OK well here's an update not sure if this helps to narrow down the problem, but I felt you should know. I just upgraded to 
Windows 10 on my Boot Camp partition, and was trying the controller out. Once I was done testing I left it plugged in, only to come back an hour later to notice my computer was still awake. I unplugged the controller and it went to sleep after the normal allotment. So basically I'm saying the fault may lie with my controller somehow or the Mac itself. I have a Rock Candy controller if this helps.

RodrigoCard, 2015-08-06:
So you are saying that it is happening on windows too  right?

GRPyles, 2015-08-07:
Yes that is correct. I also re-ran some tests with the controller plugged in on both Mac OS X and Windows. Both times prevented the system from going to sleep, once unplugged they went to sleep as normal. So my question now lies is the Rock Candy controller solely the cause? Well anyway thank you for entertaining my queries on this. I'm puzzled more now than before since I was fine leaving it plugged in on Mavericks and Windows 7 and would sleep just fine. Guess I'll just have to still unplug like I have been.

FranticRain, 2018-06-01:
The question asker found the issue to be not related the driver and I cannot reproduce this issue on a modern version of the driver, so I am going to close it.

closed 2018-06-01 by FranticRain