360Controller/360Controller #61: MadCatz GamePad (Wired) shows up in preferences pane but no input is recognized

cscheffold, 2015-01-23:
Just installed for the first time on my MacBook Pro (running Yosemite). The preference pane seems to recognize that the controller is plugged in, but it doesn't seem to recognize when I press any buttons. 

Any ideas?

[image]

thanks
Charles

RodrigoCard, 2015-01-23:
Could you test your controller with this version?
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview2.pkg

Please restart once, if you have used the driver since the last time you boot your computer.

cscheffold, 2015-01-24:
Hi Rodrigo - installed the new version, rebooted, same result. But I discovered something interesting in the process.

I updated the driver and saw that it still wasn't working, soI tried it in Parallels (Windows 7 VM) just to make sure the controller was functioning properly. It worked fine, so I disconnected the USB device from Parallels and went back to the 360Controller preference pane. The controller movements and button presses now registered just fine.

So it seems like maybe something isn't being initialized properly? I can do further tests if that helps you.

[image]

RodrigoCard, 2015-01-25:
Is your product ID/vendor ID the same as #43 ?

cscheffold, 2015-01-25:
I don't think it is the same device.

[image]

RodrigoCard, 2015-01-25:
Yeah, it is not.

Does your gamepad have some kind of switch behind? If so, changing it helps?

I cant think of anything more, the ids are correct and the driver is detecting them...
Maybe a faulty hardware?

if you can ensure that everything is working with this controller, you may need to debug the driver yourself or ask someone that have this specific controller and who knows how to do that to help you.

cscheffold, 2015-01-25:
I'm pretty sure the controller is OK, but it's not a big deal to get this working. Just thought I'd check. No worries - thanks for your help.

closed 2015-01-25 by cscheffold