360Controller/360Controller #63: List of compatible controllers

Duros, 2015-01-28:
I can't seem the find a list of compatible controllers with this driver.
Maybe it's a idea to include this in the [Readme](https://github.com/d235j/360Controller/blob/master/Readme.md) of the driver?

Pyroh, 2015-01-29:
In this [file](https://github.com/d235j/360Controller/blob/master/360Controller/Info.plist) contains the list of compatible controllers.
Feel free to make a new wiki page with these informations.

Duros, 2015-01-29:
Thanks. I'll try, correct me if I am wrong.

Duros, 2015-01-29:
Oke, I am pretty new to this. It seems like I am not able to create a custom side, I can only add that in the wiki of my own fork. Do I have to make a push request for that?

RodrigoCard, 2015-02-01:
Yes, you Can NOT pull changes on github wikis, which is a shame
http://stackoverflow.com/questions/10642928/how-to-pull-request-a-wiki-page-on-github

you need to edit the wiki of the main repository. Do you have access?

Duros, 2015-02-01:
I can't because the option is not simply there. I think only collaborators can do this, or something is checked in the setting why I can't do it. [This](http://imgur.com/N5rdYdV) is all I can see.

RodrigoCard, 2015-02-01:
I created the pages and the custom sidebar, could you check if you can edit them?

Duros, 2015-02-01:
Yes I can edit the custom sidebar, thanks.

RodrigoCard, 2015-02-01:
Nice. closing.

If you need another page, tell me here and I will create it.
Feel free to edit/add/fix any info.

closed 2015-02-01 by RodrigoCard