360Controller/360Controller #64: D-Pad shows as multiple buttons

janisc, 2015-01-30:
This has been the functionality of the driver since the beginning. D-pad shows as multiple individual buttons to the operating system. It leads to some problems with games that by default map some functionality to some buttons. It also might lead to problems with games that get confused with too many buttons. Latter is off course example of bad design in the game, rather that in the driver.

Anyway, I was hoping that maybe there could be an option where you could choose between current functionality and an alternative one where d-pad wouldn't show as bunch of buttons. I think we need to keep the current mapping as well, as many games depend on it.

My current test case is with Residual VM. Steps to reproduce, using Xbox360 wireless controller, driver release 0.14:
1) First be sure you have enable joystick support in Residual VM
2) Launch Grim Fandango
3) Move Manny with thumbstick
4) Try to interact with object in the scene

Observe that the default interaction button is mapped to one of the d-pad directions.

RodrigoCard, 2015-02-01:
there is a new functionality that will probably help you #68 , stay tuned. :)

FranticRain, 2015-02-02:
Is this Grim Fandango the original or the recently released remastered edition? Not to try and offload this bug that does need to be taken care of, but the remastered version will run on your current Mac like a charm. [Personally I'm a huge fan of GoG.com to get my games.](http://www.gog.com/game/grim_fandango_remastered)

FranticRain, 2015-02-02:
Having a look at the Residual VM source code, it looks like the hat buttons are stored as button 21, 22, 23, and 24. Currently our report descriptor doesn't have an output for button 21-24. What we could conceivably do is add an empty packet to the report descriptor at the end, and on command up, down, left, right, currently buttons 0-3 would be reported into the 21-24 range.

It feels like a bit of a hack on our end though. Really it seems like something that should be handled by Residual VM. Its using SDL. It should be able to handle this without us needing to turn this driver into even more of a piece of middleware than it already is. But of course I'm not the one running this driver, so thats just my two cents.

RodrigoCard, 2015-02-09:
does the remapping from that version I posted in #69 helps?

RodrigoCard, 2015-02-18:
@janisc Please try the remapping feature this one https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview4.pkg

RodrigoCard, 2015-02-22:
@janisc the functionality you needed to workaround this issue is present in the latest commits, closing this. :)

closed 2015-02-22 by RodrigoCard