360Controller/360Controller #66: Can't unload the kexts

RodrigoCard, 2015-02-01:
I was writing a uninstall script (#54) and noticed the kexts aren't unloading correctly. Both for Wired and Wireless controllers.
please try:
```
sudo kextunload -b "com.mice.driver.Xbox360Controller"
```
or
```
sudo kextunload -b "com.mice.driver.Wireless360Controller"
```

The controller(s) stop working as expected but the driver is still on memory
(check with `sudo kextstat | grep com.mice.driver`)

if you reload the driver it will get back working with another stance.
Apparently the old stance was never released and leak.

This is the message I got when trying to unload the kext in verbose mode, after unloading/loading a few times:
(`sudo kextunload -b "com.mice.driver.Xbox360Controller" -v 6`)
> Kext user-space log filter changed from 0xff2 to 0xfff.
Kext kernel-space log filter changed from 0xff2 to 0xfff.
Requesting unload of com.mice.driver.Xbox360Controller (with termnation of IOServices).
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Unload' request from user space.
(kernel) Can't unload kext com.mice.driver.Xbox360Controller; classes have instances:
(kernel)     Kext com.mice.driver.Xbox360Controller class Xbox360Peripheral has 7 instances.
(kernel)     Kext com.mice.driver.Xbox360Controller class Xbox360ControllerClass has 1 instance.
(kernel)     Kext com.mice.driver.Xbox360Controller class XboxOneControllerClass has 7 instances.
Kernel error handling kext request - (libkern/kext) kext is in use or retained (cannot unload).
Failed to unload com.mice.driver.Xbox360Controller - (libkern/kext) kext is in use or retained (cannot unload).

Also, this is certainly the reason why you usually need to restart (#11) when upgrading the driver.
Maybe this is related to #13 too?

Also, this may help, I think:
http://stackoverflow.com/questions/13536970/cant-unload-kernel-extension-classes-have-instances

RodrigoCard, 2015-02-01:
This might be useful too http://stackoverflow.com/questions/17345305/unload-kext-which-still-has-instances

d235j, 2015-02-02:
There's probably some resource leakage somewhere. On device unload (when controllers are disconnected), unloading the kext should be possible, but is not. I spent a little time debugging it but gave up due to lack of time; with enough effort I'm sure the leaks can be tracked down.

We probably could debug this by tracking each allocation and deallocation; something like this might work:
https://github.com/jwise/HoRNDIS/commit/659210b6661fccff5c0ef0f7e067afd3c97a6662