360Controller/360Controller #67: Settings don't load properly when the controller is connected.

RodrigoCard, 2015-02-01:
This is an old bug, as said in #65 

the user settings is not loaded properly when the controller is connected (in fact, you cant change the settings without the controller connected, so it is a common scenario)

FranticRain, 2015-02-03:
Is it possible that our preference woes are due to [The Create Rule?](https://developer.apple.com/library/mac/documentation/CoreFoundation/Conceptual/CFMemoryMgmt/Concepts/Ownership.html#//apple_ref/doc/uid/20001148-103029). Essentially, I think the issue may be that driver claims ownership of the settings file, and so the Pref Pane can't edit it.

RodrigoCard, 2015-02-03:
If so, maybe we need to ensure neither of them have its ownership.
Maybe saving the preferences in another, neutral, place is enough?

RodrigoCard, 2015-03-20:
@FranticRain any ideas on this one?

FranticRain, 2015-03-20:
I don't know for sure. We may be checking the settings too early? It works properly with my xbox one controller driver, so the core concept is sound.

RodrigoCard, 2015-12-16:
I noticed lots of third party apps that have prefpane settings in fact doesnt use the prefpane for nothing more than loading an external app for settings.

Examples: Switch Res X, Logitech Options

I guess they hit some bugs/limitations of preference panes and worked around by making an external app?

Maybe moving the preference pane contents to a separate app worth a shot...

FranticRain, 2018-06-01:
Settings bugs should be fixed now.

closed 2018-06-01 by FranticRain