360Controller/360Controller #69: Leds flashing on 0.14 release on Yosemite 10.10.2 when wireless receiver is connected thru USB-hub

janisc, 2015-02-02:
My wireless official Microsoft Xbox360 controller just blinks all four leds simultaneously when the receiver is connected thru USB-hub. The controller shows ok in the System Preferences XBox driver view and functions great in Super Meat Boy. For testing I removed all other devices from the USB-hub. The hub itself is simple 4-port hub with option to use external power source. I didn't use it and doubt it is about power.

The wireless controller and the receiver are the official Windows version of Microsoft XBox360 controller package.

I am willing to do further testing if this needed for fixing this issue.

RodrigoCard, 2015-02-03:
There was a bug with the flashing leds that was fixed a some weeks ago, maybe it is related.
I built and signed a new package with the latest commit, you can download it here:
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview3.pkg
please test and tell us if this still happens.

janisc, 2015-02-03:
The link gives me Error 404. And just to clarify: if I don't use USB-hub, in other words plug the receiver straight to my MacBook Pro's USB port, the LEDs work correctly.

RodrigoCard, 2015-02-03:
I see, must be something else... but who knows? :P
The link should be working now, please try again.

janisc, 2015-02-03:
Seems to be fixed now with 0.15Preview3. Lights work correctly with or without USB-hub.

janisc, 2015-02-03:
Actually, I am not sure anymore. Switched the controller on again and again I saw all light blinking. So, maybe it is best not to close this. I need to test some more to see, if there is a pattern.

MrOerni, 2015-05-29:
Hey. 
I couldn't get the 0.14_beta download to work with a wireless xBox 360 controller. 
But your 0.15 compilation works great!

What about updating the download link for other users?

RodrigoCard, 2015-07-10:
@MrOerni I finally got some spare time to work on this project, I hope to be able to release at last a beta of the 0.15 version soon

@janisc Is this bug still relevant?

janisc, 2015-07-10:
I am pretty sure the problem was with my USB hub after all, so I think it is not relevant and this issue should be closed. Sorry about the confusion.

RodrigoCard, 2015-07-10:
no problem! :)

closed 2015-07-10 by RodrigoCard