360Controller/360Controller #71: In 0.14 official wired controller not recognized (Yosemite 10.10.2)

mcspifferson, 2015-02-06:
All four LED lights flash simultaneously when the controller's plugged straight into my macbook but the device won't show up so I can't pick it from the drop down list of controllers. I tried the 0.15 preview in issue #69 with the same result--no devices found.

The controller is refurbished. Not sure if that has anything to do with it. Plugged it into my dad's Windows 8 laptop but the hunk of junk won't update and I have the same problem. The thing just isn't recognized, thought it seems most Windows 8 users still can't use their controllers either. I'd test it on a 360 if I had one (bought it for use on Steam and any supported emulators).

I don't know anything about this driver development stuff but I'd like to help out with the error in any way I can.

tampe125, 2015-02-07:
I have the same problem using official wireless controller.
I tried 0.15 preview, too but the issue remains.

Any suggestions?

FranticRain, 2015-02-07:
@mcspifferson You are using a wired controller, correct? Can you check and see if there is a device named "Controller" under About This Mac -> System Report -> Hardware -> USB?

@tampe125 Did you attempt to sync the controller using the sync buttons on the wireless receiver and the controller?

mcspifferson, 2015-02-07:
@FranticRain No luck. Here's a screencap.

[image]

FranticRain, 2015-02-07:
If you are using a wired controller, then the issue isn't the driver, as far as I can tell. What the computer is telling you is that no wired Xbox 360 Controller is plugged in. At least thats what it looks like when you use a bad cable with the Xbox One controller. I've never tested a wired 360 controller, so you might want a second opinion on that. All Microsoft made USB products will have a Vendor ID: of 0x045E.

mcspifferson, 2015-02-07:
@FranticRain Aw man. Thanks for the help, buddy. Luckily a replacement's getting sent in sometime soon so I may have more luck with the new one.

(On a related note I also ordered a wireless Xbone controller, which Mac users seem to be using without problem using the beta driver that was developed with that Kickstarter a while back. Regardless I'd still like to test out the new 360 controller for this driver whenever it gets delivered.)

FranticRain, 2015-02-07:
As long as you plug it in and use the beta version of this driver [or mine](https://github.com/FranticRain/Xone-OSX), it should work just as well. And without the 15 dollar price point.

tampe125, 2015-02-08:
Sorry, I meant:
_It's a wireless controller plugged in using the USB official cable_
Following your instructions I can see it in my hardware section (and with the correct Vendor ID), however the four LED are still flashing and no driver can find it (this one or yours).
FYI I am using a late 2012 iMac.
 
[image]

FranticRain, 2015-02-08:
You definitely don't want my controller driver. Its for Xbox One controllers. That won't help you one bit. But your problem is that this method will not work. Unlike the Xbox One controller, you can't turn a wireless controller into a wired one with a play and charge kit cable. There is no data transfer that occurs through that cable. You need a wireless receiver to use a wireless 360 controller with a PC.

RodrigoCard, 2015-02-09:
@mcspifferson Just be sure that your xbox 360 controller replacement is a real wired controller, not a wireless with the play and charge cable, otherwise you will need a "Wireless Gaming Receiver", Like @FranticRain said to @tampe125 . The Wireless Xbox One with a cable should work, though.

mcspifferson, 2015-02-11:
Great news! Just got the replacement 360 controller and it works perfectly in 0.14. Must've been the controller, like you said. Thanks for your help everyone.

closed 2015-02-11 by mcspifferson