360Controller/360Controller #72: Pro Ex mini not recognized device

theprojectabot, 2015-02-08:
Its the power a green see-through device.

I am using the 0.15 from that drop box link I found in the BETOP bug..

Anyways here are the specs for the controller Im trying to get going.

Pro Ex mini for XBOX:

  Product ID:	0x531a
  Vendor ID:	0x24c6
  Version:	1.00
  Serial Number:	01055480
  Speed:	Up to 12 Mb/sec
  Manufacturer:	BDA
  Location ID:	0x14500000 / 12
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

theprojectabot, 2015-02-08:
I have also tried changing the vendor id and product id to the integer representations of the above numbers in the info plist on the kext, then did a restart but no luck (I changed the PowerAMiniProEXGreen to these new product and vendor ids)

theprojectabot, 2015-02-08:
This was the link I downloaded as I saw some ProEx in the plist...https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview3.pkg

but alas no success.

RodrigoCard, 2015-02-09:
I will take a look

RodrigoCard, 2015-02-10:
Your controller ID was not on the list, I included it and built the driver, could you test?
download here:
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview4.pkg

About your Info.plist edit:
You wont be able to simply change the plist anymore because it breaks the driver signature requirements from Yosemite.
maybe it should work if you [disable the driver signature](https://github.com/d235j/360Controller/wiki/Developer-info#yosemite-and-signed-drivers), and build and load the driver yourself.

RodrigoCard, 2015-02-22:
Does anyone have this controller to help testing?
The latest preview build can be downloaded here:
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview4.pkg

if there is no feedback or if someone tell us that succeeded in using this game controller with this driver, I will close this issue soon to clean the issue list for the next release.

Cheers!

theprojectabot, 2015-02-22:
Ok I tested it and it now registers as a device on the system pref.  I wasnt able to get it to work outside of that but perhaps thats my issue :)

RodrigoCard, 2015-02-22:
well, this seems like something game-related. where exactly you are not able to use it.

Thanks for testing anyway.

closed 2015-02-22 by RodrigoCard