360Controller/360Controller #74: Add new controller

rubiesonthesky, 2015-02-08:
I'd like to kindly ask you could add this controller to your list. It is nameless but it was popular (and blue) on ebay where I bought it. 

Controller
  Product ID:	0xf301
  Vendor ID:	0x046d  (Logitech Inc.)
  Version:	1.10
  Serial Number:	102CE9F
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft Inc.
  Location ID:	0x14240000 / 15
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

[image]

RodrigoCard, 2015-02-10:
I added your device, please test it: ;)
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview4.pkg

Cheers!

rubiesonthesky, 2015-02-11:
It works so nice now! Thank you!

closed 2015-02-11 by RodrigoCard

GoshDamn, 2015-04-19:
you are my hero and i love you

RodrigoCard, 2015-04-21:
@GoshDamn You should love everyone that contributed to this project, really :)