360Controller/360Controller #76: Rock Candy controller not detected (new device?)

LassiHaaranen, 2015-02-14:
Rock Candy controller is not detected at all it only blinks once when plugged in. The product ID doesn't seem to match the current Rock Candy controller one. Tried on OS X 10.9.5, details:

    Rock Candy Gamepad for Xbox 360:
      Product ID: 0x021f
      Vendor ID:  0x0e6f
      Version:   1.00
      Serial Number:  00021162
      Speed:  Up to 12 Mb/sec
      Manufacturer: Performance Designed Products
      Location ID:  0x04100000 / 4

RodrigoCard, 2015-02-22:
Hello @LassiHaaranen  , I added your device ID, Could you test this build, please?
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview5.pkg

LassiHaaranen, 2015-02-23:
Hi, seems to work perfectly. Thank you!

closed 2015-02-23 by RodrigoCard

fourthseed, 2017-07-11:
Hey @RodrigoCard  thanks for the post. Tried the dropbox link, but received 404 error. Would you mind resending the .pkg? Thank you.

RodrigoCard, 2017-07-11:
@fourthseed 
dropbox disabled all the public links, please download the latest version here instead https://github.com/360Controller/360Controller/releases