360Controller/360Controller #77: not working Afterglow Gamepad for Xbox 360 - new VendorID

timnhoffmann, 2015-02-15:
Not working:
Afterglow Gamepad for Xbox 360:

  Produkt-ID:	0x0413
  Hersteller-ID:	0x0e6f
  Version:	1.00
  Seriennummer:	06923F48
  Geschwindigkeit:	Bis zu 12 MBit/s
  Hersteller:	Performance Designed Products
  Standort-ID:	0xfa130000 / 6
  Verfügbare Stromstärke (mA):	500
  Erforderliche Stromstärke (mA):	500

The product id is not in the current list of supported devices. I checked that it will work by temporarily disabling the kext signing and then using DriverTool to add the a "Controller 3695 1043".
Btw: after rebuilding the kextcache and then reactivating the signing it still works - until the system decides to redo the cache once more I guess...
Can it be added to the signed driver at some point?
best,
T.

RodrigoCard, 2015-02-18:
I will add that soon

RodrigoCard, 2015-02-22:
Hello @timnhoffmann , I added your device ID, Could you test this build, please?
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview5.pkg

timnhoffmann, 2015-02-23:
Dear Rodrigo,

Thank you. I tried the build (tried to throw out the old one first to be sure). It works for me but only with kext-singning turned off ("otherwise I get "com.apple.kextd[19]: ERROR: invalid signature for com.mice.driver.Xbox360Controller, will not load" in the Syslog), so it is not signed I guess?
The rumble, that was linked to the shoulder knobs in the old PrefPane is gone (I don't know wether on purpose or by accident - I have no testcase that would use them...).
The new PrefPane looks good though :)

best,

Tim

RodrigoCard, 2015-02-23:
Oh sorry, I have a couple of sign keys with similar names for different purposes that I cant rename to make easier to remember which is the one I need to use in each situation. And everytime I pull a commit I need to remember to double check that... sigh

Also, the Forcefeedback seems broken...

Here is the fixed build with the correct signature:
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview5.pkg
(yes, it is the same link, I just replaced the file)

timnhoffmann, 2015-02-23:
Really no need to appologize! Thank you for doing this :) 
Now it just works as expected!

all the best,

Tim

closed 2015-02-23 by RodrigoCard