360Controller/360Controller #78: Add new Device: Speedlink Strike FX Wireless

leukhin, 2015-02-16:
Hello, 
Is it any chance Speedlink Strike FX Wireless to be added?
I have tried to add it by myself to Info.plist, but no luck.

The issue is this is wireless controller with radio receiver. And it is visible on USB section on mac, but when I power on controller it cannot establish connection to receiver. I don't know is this request applicable to 360Controller. If not - just close this issue. 

I'm googled a lot for solution and I didn't found any approach.

This is how controller looks:
[image]

This is my USB info: 
[image]

Thanks a lot!

RodrigoCard, 2015-02-18:
Hello,
This driver was made for Xbox and Xinput devices in mind, not regular HID controllers.
Sorry

closed 2015-02-18 by RodrigoCard

leukhin, 2015-02-18:
@RodrigoCard Yes, no problem.

Thanks.