360Controller/360Controller #79: Is it possible to remap buttons/axis?

weltonrodrigo, 2015-02-18:
I play this game called Crimsonland.

The gamepad works ok on Steam Big Picture and other games, but this one no.

It seems to get axis wrong. Right trigger seems switched for right analog sticker.

Is there a way to remap buttons or axis so I can trick this game?

Thank you.

FranticRain, 2015-02-18:
I recently added a feature that will let your rebind buttons. Haven't gotten around to making the axes remappable. This will remap the controller completely, not just from Crimsonland. Try this build and let us know how it works out. https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview4.pkg

weltonrodrigo, 2015-02-18:
Thank you so much for sending this build. The pref pane is great and the rebinding seems fine. But, unfortunately, the problem in Crimsonland are the axis, I think. 

Checked better and the problem is: left trigger is mapped to x-axis and right trigger is mapped to y-axis. Both on the right stick.

smoketetsu, 2015-03-20:
Since it's to have button/axis rebinding.. Will this driver have profile capability for different apps?

FranticRain, 2015-03-20:
Since the driver is working at a very low level, (it is basically loaded as part of the operating system) it cannot tell who or what is using it. So from the driver's perspective, there is no way to do this.

smoketetsu, 2015-03-20:
Well, even if you can't make it so it can automatically detect the game and apply a profile how about saving the layouts to a user definable name so that way one doesn't have to reconfigure the buttons\axis every time they want to use certain games and instead can go into the pref pane and change to a saved layout? Most games work well with the default layout so this would probably be only necessary for some edge cases.

FranticRain, 2015-03-20:
I'll look into it.

RodrigoCard, 2015-03-20:
I agree with @FranticRain , this kind of stuff doesnt belong to the prefpane and driver itself.
Maybe a separate app?

MrLavender, 2015-06-27:
The GTA 3 series of games also have this problem with using the L/R trigger axes instead of the right analog stick. It's hard-coded into the game (probably just picks the first axis it sees or something) and so makes it impossible to play because it takes the trigger axis zero readings as the stick being at full up+left, and you can't disable or change it.

I just added a small hack to make the triggers report the same reading as the right analog stick. Not a very elegant solution but it works perfectly for my needs. Full axis remapping would obviously be much better but something like this at least fixes the dumb games that think the triggers are the right analog stick.

```
void Xbox360Peripheral::fiddleReport(IOBufferMemoryDescriptor *buffer)
{
    ...axis inversion and deadzone fiddling...

    if (gta3hack) {
        report->trigL = report->right.x/256 + 128;
        report->trigR = report->right.y/256 + 128;
    }
}
```

smoketetsu, 2015-06-28:
MrLavender I suggest you look into Ginput mod for GTA3 series. Before you wonder, yes.. it'd work. The GTA series on Mac is the Windows version in a Windows game wrapper... http://www.gtagarage.com/mods/show.php?id=24272 

There's also this one for GTA:SA: http://gtaforums.com/topic/427517-rel-xcontrols-xinput-based-gamepad-support/

I used these mods to enable better X360 controller support in my GTA:SA and played from star to finish that way.

xythobuz, 2015-07-26:
+1 for remappable axes, I would love to have this feature!

RodrigoCard, 2015-08-28:
the buttons are remappable for some time already, closing this one

closed 2015-08-28 by RodrigoCard