360Controller/360Controller #80: Broken force feedback in prefpane?

RodrigoCard, 2015-02-23:
I am not sure if I did something wrong or if something changed recently break it, but the rumble doesnt seem to be working for me in the preference pane anymore.

@MaddTheSane can you confirm that?

FranticRain, 2015-02-23:
Doesn't work for me either. I probably broke it at some point. Time to bust out the blame tab!

MaddTheSane, 2015-02-23:
Force feedback in games seem to work.

Just an FYI: I've been sick the last few days, so I haven't done much but sleep. I'm starting to feel better, though. So by Tuesday I should be in good enough health.

RodrigoCard, 2015-02-23:
@FranticRain it is fine! No blame needed :)
@MaddTheSane I hope you get better! Take your time.

smoketetsu, 2015-03-20:
Speaking of Force Feedback in certain games such as force feedback enabled Wine games it seems to be rather sloppy as to when force feedback happens and it seems to keep on going sometimes until either another event interrupts it or I close the game.

RodrigoCard, 2016-01-19:
it is working for me since the latest updates. does anyone still have problems with this?

closed 2016-05-16 by RodrigoCard