360Controller/360Controller #81: some help required for the 360 controller drivers

el-bo, 2015-03-08:
hi

just wondering if i could get some help with some issues

i have been using various iterations of the tattiebogle driver over the last 6 years with no issue. i have been slowly migrating software over to a yosemite partition, and making sure all is ok. most of the games i use are working fine with your version of the driver, but i am having issues with others

my main issue is with a game called 'metro: last light'. unfortunately the controller is greyed out on the menu despite the controller being compatible. currently involved in an ongoing support discussino with the developer, who has referred me to the tattiebogle drivers. i have yet to mention that i am using these updated drivers for yosemite. it might be a stretch to ask if anyone else has this particular game, but maybe there are known issues with certain games, for certain reasons. if i can't get it working, i might have to pursue a refund. i do really want to enjoy this game though :)

my main reason for writing, however, is to see if there is anyway of streamlining the uninstall process. i am sure everybody is familiar with conflicts between the tat drivers, and companies that write proprietary drivers e.g aspyr. searching for pref files, removing, and restarting is the advice given by aspyr, but i tend to jump from one game to another in quick succession; it is not possible to keep going through this process for a quick 10-minute session

i have to say that this coding stuff goes way over my head, and i am often quite wary of making modifications (i don't even know what i need to do with the uninstall script above), but i was wondering how simple the process can be made, and if there is any way of avoiding a restart. i did come across this the other day, but i am not sure whether it applies to this situation :

http://onlivefans.com/news/xbox-360-controller-issues-with-onlive-on-a-mac-can-be-fixed/357/

one other solution might be to buy 'controllermate'. that way, as long as there is a current functioning driver, it would be possible to bypass the controller options, and defer to emulating mouse and keyboard bindings. this does seem to be a huge amount of work though, even as a last resort

this is my very first github post, and i am hoping this whole controller thing becomes a bit less 'fringe'. as someone who has had both wrists operated on for carpal tunnel (and can only game with a controller, in shorter bursts), i am so grateful for everything you are doing

any help would be greatly appreciated

many thanks

duozmo, 2015-03-08:
Welcome to GitHub! I second your request for an uninstaller.

I don’t have experience with Metro: Last Light, but the [App Store reviews](https://itunes.apple.com/us/app/metro-last-light/id665063218?mt=12) make it sound like the Mac port is very buggy. It may not be possible for this project to offer controller support for the game. Perhaps if [Redux](http://store.steampowered.com/app/287390/?snr=1_7_15__13) [[?](http://www.polygon.com/2014/8/8/5983833/metro-redux-trailer-2033-last-light)] gets released for Mac, it can happen.

I’m no Github veteran myself, nor a contributor to this project, but I would suggest two improvements to your post:

1. Keep each Issue narrowly focused
2. Use more descriptive subject lines

“Issues” aren’t great for multifaceted conversations. In this case, I would have started separate Issues for an uninstaller and bugs with Metro: Last Light.

RodrigoCard, 2015-03-09:
Hello,

If the Metro Game supports xbox controllers without driver, the commands from the link you provided should work to disable this driver.

That said, the option to disable the driver was already requested before, see here: #21 

I will take a look into that.

FranticRain, 2015-03-09:
Ninja'd by Rodrigo! Dang!

There is already an existing issue regarding the uninstaller, #54 so lets not duplicate that. When the issues its facing are resolved, likely there will be a file named "uninstall.command." To uninstall, a user will simply need to double click on it, enter their password at the Apple prompt, and then the driver will be uninstalled. However. What the OnLive article is suggesting is not an uninstall, it is an unload. Simply put, what it does is it hides the driver from OS X, so the operating system does not use it. Then when you load it again, it can be used by OS X once more. It is important that this requires administrator privileges to do both of these tasks, since you are messing around with the operating system.

I can quickly put together a load.command and an unload.command that you can keep on your desktop or some such place to do this if you would like. I'm not very well versed with the menu-bar/daemon part of the driver, so someone else would have to implement a more permanent fix.

el-bo, 2015-03-09:
thanks guys !! really appreciate the input :)

------------------------------------------

@duozmo : haha, i guess it was a bit of a rambling post. i guess it starts with an attempt to get to the bottom of the 'metro' compatibility issues, which then led to the search for a more elegant uninstall method.

it is true that the mac port of the game seems to come with issues, but i am not sure that the controller issue is one of them. certainly, the developer (with whom i have an open support ticket) seems to think it should just be a case of installing the tattiebogle drivers. this lead me to think that there is something in this latest fork that might be causing a conflict (though all the other games i have tried thus far have worked without hitch). either way, i thought it might be a good idea to remove all traces of the driver and start from the beginning

i will try to get a little more familiar with navigating this place, and then attempt to keep my entries a little clearer

thanks

------------------------------------------------
@RodrigoCard 

thanks :)

i will have a try with those commands when i have a moment. it would be great to try and get this as a feature though; as i mentioned, aspyr are a company who have their own proprietary drivers that cause conflicts. i don't currently own their games for this very reason, but there are some titles i would like to buy. 

thanks

-------------------------------------------------

@FranticRain 

thank you :)

i had a look at the link to which you referred, but most of this stuff goes right over my head :)

hiding the drivers sure sounds like a more elegant option, but it depends on how easy it is for you guys to implement

i can't really offer any help, but i have gratitude to spare :)

thanks again guys

FranticRain, 2015-03-09:
Here are some scripts. They only work on 10.9+ because I've hard coded the /Library/Extensions location. Hopefully they work for you!
https://dl.dropboxusercontent.com/u/33409904/load.zip
On more of a housekeeping note, if we continue to use these, we probably need to move them into the project dmg and off of my Dropbox.

el-bo, 2015-03-09:
@FranticRain 

cheers man. had a go with these scripts on my yosemite partition. they do work as advertised, but no luck with 'metro' (unloaded or reloaded). evidently they don't have a proprietary driver (which is why they recommend the tat driver), so there seems to be some issue with this new driver or their controller support in yosemite

my support ticket with them seems to have gone cold, so i might just have to try for a refund

the scripts worked perfectly when trying with 'driver', so hopefully this bodes well for the aspyr issue. now i just have to look out for an aspyr game on sale. of course, if anyone else can check the scripts against aspyr's offerings that would be great

thanks again for all the help. i will keep you posted on any further discoveries

MaddTheSane, 2015-03-11:
Sorry I've been silent: I've been going through some upheavals in my life.

Right now, I'm looking into adding a menu item to turn off wireless controllers, and trying to figure out if it can be part of the daemon (it doesn't look like it).

RodrigoCard, 2015-03-12:
People: I made some ui changes (somewhat unrelated) and was thinking where I should put a button to disable the driver, please take a look and share your thoughts: #82

el-bo, 2015-03-17:
hi guys

i have been following parts of this unfold via my emails. it seems this has branched off to other areas, and i would rather leave those for more technical aspects of the process

i just wanted to offer my continuing thanks for what you guys are doing here

all the best

mele113, 2015-05-01:
Thanks to all whom made tis controller driver work for me!

RodrigoCard, 2015-07-28:
Latest version can be disabled without uninstalling
https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial

closed 2015-07-28 by RodrigoCard