360Controller/360Controller #83: Can't get Xbox360 controller to appear in PrefsPane

sansgil, 2015-03-15:
I've installed v0.14 on Yosemite 10.10.1. 

I had installed an earlier version of the drive from tattiebogle though I had never had an Xbox 360 controller working on my mac.

So I completely removed all references to the kexts and prefs files from my mac (detailed below) and then reinstalled the driver

/Library/Application Support/MICE/360Daemon
/Library/LaunchDaemons/com.mice.360Daemon.plist
/Library/PreferencePanes/Pref360Control.prefPane
/System/Library/Extensions/360Controller.kext
/System/Library/Extensions/Wireless360Controller.kext
/System/Library/Extensions/WirelessGamingReceiver.kext

Still every time I connect my controller via USB cable, it comes up with the same message dialog saying that the pad is charging but wont work without an official Xbox wireless receiver.

Can you shed some light on why I can't seem to get this driver to work correctly? There's nothing obvious in terms of errors in Console>All Messages.

FranticRain, 2015-03-16:
Xbox 360 controllers aren't as cool as you might think. As it turns out, a wireless Xbox 360 Controller ONLY works wirelessly. And a wired Xbox 360 Controller ONLY works wired. So when you plug a play and charge kit into an Xbox 360 Controller, and then plug it into the computer, the Xbox 360 Controller isn't working, because its not working wirelessly. A wireless Xbox 360 Controller can only be used with a computer using a wireless adapter. No amount of cables is going to solve that problem.

So the driver won't work correctly because the controller isn't sending any information. Its just charging.

However, Xbox One Controllers are super cool and will actually switch between wireless and wired modes based on USB cable connectivity. Its an awesome feature.

sansgil, 2015-03-16:
@FranticRain Thanks for the speedy reply.

So as I understand it I'd need to buy an Xbox 360 wired only controller of an Xbox One controller in order to get this to work?

FranticRain, 2015-03-16:
Or a wireless adapter for the Xbox 360 Controller. Xbox One controller support is around in pre-release builds, but isn't slated for official release until v0.15. [Heres one of those wireless adapters if you're looking for one.](http://smile.amazon.com/Wireless-Gaming-Receiver-Xbox-360-xbox360/dp/B0076HD2W8)

sansgil, 2015-03-16:
@FranticRain Thank you.

closed 2015-03-16 by sansgil