360Controller/360Controller #84: Green leds flashing on 0.14 release on Yosemite 10.10.2 with wired controller

gustaflindqvist, 2015-03-17:
I having problem with the latest release and my new Xbox 360 wired usb controller. I getting green leds flashing as a ring and the controller won't show up as device in the preference.

I can see the controller in the system report:
'```
Product ID:	0xf301
Vendor ID:	0x046d  (Logitech Inc.)
Version:	1.10
Serial Number:	1031540
Speed:	Up to 12 Mb/sec
Manufacturer:	Microsoft Inc.
Location ID:	0x14300000 / 1
Current Available (mA):	500
Current Required (mA):	Unknown (Device has not been configured)
```

Is there something I missing or what?

gustaflindqvist, 2015-03-17:
I can see in the console:
```2015-03-17 19:21:45,709 System Preferences[13076]: No devices found? :( device count==0, i==0```

RodrigoCard, 2015-03-18:
Could you test this version?
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview5.pkg

gustaflindqvist, 2015-03-18:
@RodrigoCard Woho! It's working with version 0.15 Preview 5! New Vendor for the controller in the preview version?

RodrigoCard, 2015-03-18:
Yes. We added lots of new devices. 0.15 will be pretty inclusive :)

closed 2015-03-18 by RodrigoCard

gustaflindqvist, 2015-03-18:
Awesome. Great work!

I found the source code where I could have added the device: https://github.com/d235j/360Controller/blob/master/360Controller/Info.plist

Doh!

RodrigoCard, 2015-03-18:
Just editing the plist file will not work unless you disable the Yosemite's kext signature checking. :)