360Controller/360Controller #85: Couple of (Wine) games crashing with latest 0.15 preview.

smoketetsu, 2015-03-20:
The games in question are Mirror's Edge and Alan Wake.

I tried disabling Force Feedback support in x360cemu in my Mirror's Edge wrapper thinking that might have something to do with it and that didn't help. I also tried changing the Wine engine because some other UE3 based games such as X-Men Origins Wolverine weren't crashing. 

The only thing that helped is downgrading to the 0.12 drivers from Tattiebogle.

RodrigoCard, 2015-07-28:
Hello!
does this happen with this version?
https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial

smoketetsu, 2015-08-04:
Since I ugpraded to Yosemite yesterday (to play shadow of mordor). I also of course had to upgrade my controller driers. 

Those drivers did cause Mirror's Edge in Wine to fail to start as well. It gets up to the spinning icon that shows that the game is loading and then the game closes. 

Also, I seem to be noticing a negative performance impact on some of my games such as LEGO Jurassic World, Serious Sam 3, Talos Principle with the 0.15 driver that the 0.14 version doesn't seem to produce. The games are visually less smooth and responsive.

smoketetsu, 2015-09-03:
Downgrading to Mavericks helped with the performance issues but games such as Mirror's Edge in Wine still crash with the latest drivers from here.

smoketetsu, 2015-11-27:
I've recently upgraded to El Capitan... 

So far so good on the performance front. In fact the performance seems noticeably better in El Capitan for me than in Mavericks unlike what happened with certain games in Yosemite vs Mavericks. Even things that where unaffected before look and feel noticeably smoother in El Capitan. 

However, the controller drivers from here the newer versions of OS X and my wireless 360 controller is still causing some of my UE3 games in Wineskin like Mirror's Edge and Wolverine to fail to load and rumble to malfunction in games like DMC4 in WIneskin where it starts rumbling when I get hit and continues rumbling for a while on its own where it should have only rumbled during the hit.

FranticRain, 2018-06-01:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-06-01 by FranticRain