360Controller/360Controller #86: Controller only connects briefly and flashes

semse, 2015-03-25:
Hi,

I have not seen this in any of the other issues.
I have a wireless xbox360 gamepad and tried 0.14 and 0.15 (pre-release)

Description:

- Plugged in receiver, switch on controller (works the other way around too)
- controller is recongnized, player1 light turns on
- after 5secs, LEDs start spinning on controller but it is still working
- after around 20secs, controller just turns off, even if it's midgame and i am using it (sometime, it just turns off after 1min, sometimes even after 5seconds)
- happens with preferences pane started or not

Thanks!

Contez, 2015-04-09:
Might be a dumb suggestion but worth checking.

My controllers does the same thing to me... when the batteries are nearly dead. ;)

Have you tried with fresh batteries?

semse, 2015-04-09:
I tried on a windows computer. Everything works fine on that one.

RodrigoCard, 2015-04-10:
if you updated, reinstalled, or something like that, be sure to restart your mac, sometimes it does not reload the drivers properly in these cases.

Naf3tS, 2015-04-20:
I have the same problem. 10.10.3 + 0.14. With another controller it connects but the light on the controller keep 'spinning'.

RodrigoCard, 2015-07-28:
This appears to be related to not restarting the computer after installing the driver, which is needed because of the bug #66 

Anyway, install the latest version to be sure
https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial

Closing this one.

closed 2015-07-28 by RodrigoCard