360Controller/360Controller #88: Thrustmaster GPX Gamepad

FactSphere, 2015-03-28:
Would you be so kind to add the following device?

GPX Gamepad:

  Product ID:	0xb326
  Vendor ID:	0x044f  (ThrustMaster, Inc.)
  Version:	4.02
  Serial Number:	00000001
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Thrustmaster
  Location ID:	0xfa130000 / 5
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

Thanks!

RodrigoCard, 2015-04-02:
sure, I will add that soon.

RodrigoCard, 2015-04-17:
Hello, I just built a new pkg with some new devices included, please install, restart and test.
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview6.pkg

Cheers!

FactSphere, 2015-04-18:
It works perfectly, finally I can play "I am Bread" on OS X :D Thank you!

closed 2015-04-21 by RodrigoCard