360Controller/360Controller #90: Xbox 360 racing wheel

codecowboy, 2015-04-07:
Has anyone tested this with the Xbox racing wheel + pedals?

RodrigoCard, 2015-04-10:
No. Do you have one? If it is not working, maybe it is just a matter of adding its vendor/product ids to the info.plist.

codecowboy, 2015-04-10:
I do but its wireless so would need to buy the receiver to see if it works
and not sure if its worth it if the driver won't work. Its this one:

http://en.m.wikipedia.org/wiki/Xbox_360_Wireless_Racing_Wheel

RodrigoCard, 2015-04-17:
we will be glad to add this device IDs once somebody can send it to us and test,
we just cant guarantee it will work properly before that.

If you (or someone else) get the Receiver and the controller together, please reopen this issue and include the device IDs.

Thank you!

closed 2015-04-17 by RodrigoCard

MrSnowflake, 2016-12-10:
How can I get the device id of the wireless wheel? I have the original Xbox 360 Wireless wheel.

deansallinen, 2017-04-12:
I also have the Xbox 360 Racing wheel + Pedals and would like to see support included! How do I find the device ID?