360Controller/360Controller #91: PDP Marvel Wired Controller

Barryrowe, 2015-04-09:
Please consider adding the following device in the next release:

Product ID:	0x0147
 Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	06DC7A30
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14100000 / 2
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

I attempted to add this to my local installed plist file with the following entry, but did not have luck. Is there more that needs to be changed?

```xml
<key>PDPMarvelXboxOneController</key>
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.mice.driver.Xbox360Controller</string>
			<key>IOCFPlugInTypes</key>
			<dict>
				<key>F4545CE5-BF5B-11D6-A4BB-0003933E3E3E</key>
				<string>360Controller.kext/Contents/PlugIns/Feedback360.plugin</string>
			</dict>
			<key>IOClass</key>
			<string>Xbox360Peripheral</string>
			<key>IOKitDebug</key>
			<integer>65535</integer>
			<key>IOProviderClass</key>
			<string>IOUSBDevice</string>
			<key>idProduct</key>
			<integer>327</integer>
			<key>idVendor</key>
			<integer>3695</integer>
		</dict>
```

MaddTheSane, 2015-04-09:
Due to Apple enforcing kext signing, you would have to build the driver from source.

That said, this could easily be added to the existing property list in GitHub.

closed 2015-04-09 by MaddTheSane

Barryrowe, 2015-04-10:
Thanks. For those of us without a Apple Developers license, and not wishing to disble code signing in Yosemite, is there an expected release time for the next build?

RodrigoCard, 2015-04-10:
@Barryrowe pretty soon, I just want to finish #82 and some other minor stuff...
if everything is al right I will sign and release.

In fact I was planning to do that the last week, but the work (and some gaming, I admit) got in the way.
time to stop procrastinating! :)

RodrigoCard, 2015-04-17:
@Barryrowe 
Hello, I just built and signed a new pkg with some new devices included, if you want to test it, you can download it here:
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview6.pkg

Cheers!

Barryrowe, 2015-04-18:
Looks good! It's picking it up now.

commit 94dfeec5 (2015-04-09): Add PDPMarvelXboxOneController

Fixes #91