360Controller/360Controller #92: Help with deactivating vibrations

BCMNG, 2015-04-09:
Hey there,
I was searching for updated versions and was lucky enough to land here.

I use a wired XB360 controller and play almost only CS:GO (I know, I know... :) ).

Apart from some hickups with the movement of the sticks, the only thing really giving me a hard time is the vibrations. I tried it with re-mapping and using different buttons for shooting - still - just doesn't feel right. So I was hoping that someone here can maybe help me out.

I have 0 experience with commands/terminal. Is there a way to deactivate the vibrations without taking the controller apart and without messing up my mac.

Don't want to sound stupid. I really don't have any clue how to change it.

And thank you all for keeping this alive!

MaddTheSane, 2015-04-09:
The game itself should have a way of disabling force feedback, either in game or controller settings.

BCMNG, 2015-04-09:
There isn't. I made screenshots from everything
* Steam
* Steam Big Picture
* Counter Strike: Global Offensive (in game)

I can't find it. Maybe I'm looking at the wrong place?
Google/Steam forums - no luck either.

Here are the pictures - http://imgur.com/a/a7zRs
Let me know if the link doesn't work.

BCMNG, 2015-04-15:
Can be closed. Opened the controller and took the rumble packs out.

Deph0, 2015-04-15:
Maybe we could implement this next to the disable driver options in #82 ?

Contez, 2015-04-15:
I agree with Deph0.
It could be useful.

RodrigoCard, 2015-04-17:
@Contez @Deph0 @BCMNG 
This option will be added soon, Thanks to @FranticRain .
See #94 
Closing this one.

PR #94 (2015-05-19): Add rumble disable
I changed "Xone Rumble Options:" to "Rumble Options:" For non-Xbox One controllers, you are given the options "Default" and "None." If you are using an Xbox One controller, it also gives you "Triggers Only" and "Both." I've been working on my driver, so I can't test it, but all of the code is super straight forward.

Also I snuck in a new Xbox One controller. For some reason, my driver doesn't work with PowerA controllers, but this one might? I'd like to get an installer for this version once it is merged in order to test the additional controller support.

EDIT: Oh yeah, re [... cut]
closed 2015-04-17 by RodrigoCard