360Controller/360Controller #93: RockCandy Wired controller

hollanolla, 2015-04-10:
I've asked about this on multiple sites to have no help at all. I've downloaded every patch going and every driver but nothing seems to work. I plug in my wired rock candy xbox 360 controller and it lights up for a second and then nothing. It isn't detected at all and just doesn't work. Any help from here?

Barryrowe, 2015-04-12:
Do you have the same version that is in the current .plist file? To check:

1. Plugin your controller to the USB
2. Click the Apple Menu
3. Click About this Mac
4. Click the "System Report..." button
5. Select the USB
6. Find the Controller in the list on the right-top
7. Find the VendorID and ProductID values for the Controller

It should work if your productID is one of the following:
ProductID:  0x011f
ProductID:  0x0146
ProductID:  0x021f

and your vendorID is:
VendorID:   0x0E6F

If your values are different, the version of your controller may not be listed in the set of supported controllers. You can post the values on here and it should be able to get added.

hollanolla, 2015-04-12:
Hi Barryrowe! Thanks for the reply, I've checked and my controller's product ID is 0x021f and the vendor Id is 0x0e6f so it should work! But alas it doesn't. Any other ways around this?

RodrigoCard, 2015-04-13:
try this https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview5.pkg , restart, and tell me if that works.

hollanolla, 2015-04-13:
@RodrigoCard oh my god it works!!! Thank you so much!!!!

closed 2015-04-13 by hollanolla

slytherclawgamer, 2017-10-17:
My VendorID is 0x24c6, and my ProductID is 0xfafe. Is this supported? I ask because my controller is not presently working.

OsbourneCox, 2025-01-21:
> [#76](https://github.com/360Controller/360Controller/issues/76) this one have the same id
> 
> try this https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview5.pkg , restart, and tell me if that works.

Link is broken. Long shot this gets seen, but what was the resolution here?