360Controller/360Controller #95: Mad Catz FightStick TE2

DrOMFG, 2015-04-15:
Hello,
Is support for this stick possible?
Thank you :)

Mad Catz FightStick TE2:

  Product ID:	0xf080
  Vendor ID:	0x1bad
  Version:	4.90
  Serial Number:	1045A766
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Mad Catz, Inc.
  Location ID:	0x14230000 / 11
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

RodrigoCard, 2015-04-17:
Hello, I just built a new pkg with some new devices included, please install, restart and test.
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview6.pkg

Cheers!

DrOMFG, 2015-04-17:
I did some testing and everything seems to be working now. Thanks!

RodrigoCard, 2015-04-17:
thanks for testing!

closed 2015-04-17 by RodrigoCard