360Controller/360Controller #96: Logitech F310 Left stick and Dpad reversed

darthcircuit, 2015-04-16:
Hey, I recently got this set up on my wife's macbook with Yosemite and Logitech F310 Controller. The controller works ok other than the fact that whenever she moves the left stick it selects a direction on the D-Pad, and when she presses a button on the D-Pad, it selects a direction on what would be the left stick. Is there any way to reverse or remap this?

FranticRain, 2015-04-16:
Does the controller exhibit this functionality is just games or also in the preference pane? You should be able to test with the controller while the preference pane is open. If the problem is not present in the preference pane, it is likely that your issue is with the bindings of the game. You will have to deal with that on a per-game basis. However, if it still happens in the preference pane, something extremely odd is going on.

Simply put, the D-Pad is a digital input. It takes up a single piece of space. A joystick is an analog input. It takes up 16 times that amount of space. So if the packet from the controller is messed up, that would involve taking a binary state (0 or 1) and turning it into a number between -32768 and 32767. We would have to create detail which doesn't exist.

So test the controller in the preference pane, and let me know what that looks like.

darthcircuit, 2015-04-16:
We first discovered the problem when we were playing borderlands 2, so we hopped into the preference pane and it was happening there too. When the stick gets moved, it shows a direction on the dpad. when pressing the dpad, it seems to go to the max value on whichever axis is selected. strange...

FranticRain, 2015-04-16:
Ah! I looked at the documentation for the F310. You'll find there there is a "Mode" button on your controller. It's feature is documented as such:

"The Mode button on your Gamepad F310 swaps the functionality of the directional pad (D-pad) with the left thumbstick. When the Mode button is lit, the functionality is being swapped.

If either your D-pad or thumbstick isn't operating as expected, press the Mode button. This should resolve a situation where the D-pad is acting like a thumbstick or vice versa."

So press the mode button and see what happens!

darthcircuit, 2015-04-16:
I will look into that :) thanks so much for your help :) i'm helping my wife diagnose this over skype lol. I'll get back with you as soon as i can :)

RodrigoCard, 2015-05-19:
Looks like you got it working, I will close this issue. If you still have this problem, please reopen. Cheers!

closed 2015-05-19 by RodrigoCard