360Controller/360Controller #97: release a signed compiled beta version of 0.15

slawo, 2015-04-23:
Could you start releasing signed beta versions of the driver?

RodrigoCard, 2015-04-23:
you can download one from last week here: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview6.pkg

mele113, 2015-04-28:
I'm confused.as a noob should I ope 0.14 or 0.15 I downloaded both and am scratching my head
[image]

xxfree86, 2015-04-29:
my logitech f710 controller now works with this beta driver.  great works!

Minopolis, 2015-05-03:
I am amazed. I've been trying to get my f710 to work in xinput mode for ages and you made it work. Thank you so much. Where can I donate? :smile:

mele113, 2015-05-04:
I used 0.15 new version and it works perfectly with Yosemite  xbox360 wireless controller with reciever from microsoft

RodrigoCard, 2015-05-11:
People, just a update, I got really sick this week, I was planning to check the last pending tasks and release, but it is hard when you are about to vomit everytime you look more than 5 minutes at a screen. :/

If someone can help with stuff here, I would appreciate.

RodrigoCard, 2015-06-01:
just a new signed build, while no beta is available. https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg Cheers!

RodrigoCard, 2015-07-10:
Okay, I have some free time finally.
Lets squash some bugs/tasks and release at last a beta of 0.15....

RodrigoCard, 2015-07-10:
Important before releasing the beta: #115 #101 
I think the others https://github.com/d235j/360Controller/milestones/0.15%20version can be left to a final 0.15 release.

RodrigoCard, 2015-07-28:
DONE!
https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial

Sorry for taking too long!

There are a few important issues here that need to be fixed before a proper release https://github.com/d235j/360Controller/milestones/0.15%20version but this one is usable :)

@d235j @MaddTheSane @FranticRain 
Also, Tattlebogle is back  https://github.com/electric-monk/360Controller/
Lets see If he is willing to accept a big pull request with all this work :)

closed 2015-07-28 by RodrigoCard