360Controller/360Controller #98: Uninstalling the driver

rmaob, 2015-04-26:
How does one uninstall this? Need to uninstall to resolve some conflicts before reinstalling

IcyTesseract, 2015-05-10:
This command should tell you the files related to 360Controller:

```
pkgutil --files com.mice.pkg.Xbox360controller
```

Just copy and paste into the Terminal

Deph0, 2015-05-10:
There will be an uninstall button inside the preference pane in the new version; as per issue #54.

julian-weinert, 2015-05-23:
I don't get any files returned by `pkgutil`.
Can you maybe provide a list of files gettimg installed?

RodrigoCard, 2015-05-23:
this is the command you need:
````
pkgutil --files "com.mice.pkg.Xbox360controller"
````
also, try this driver version first
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview6.pkg
it can be disabled temporarily and includes an uninstaller.
(you may need to restart once after installation)

BCMNG, 2015-06-20:
3-4 days ago I wanted to re-install the driver as I'm having issues with a specific game (since their last update).

I'm not able to uninstall:
[image]
[image]

Additional note:
Before installing this ...preview.pkg version (weeks ago), I did not manually uninstalled the previous version. Could this be the reason for the error?

in case the upload doesn't show:
http://i.imgur.com/Uvrg6xM.png
http://i.imgur.com/txlKZRT.png

RodrigoCard, 2015-06-23:
Enabling and disabling is not that reliable right now.

Please do this:
* remove all xbox hardware connected to your mac
* restart your mac
* uninstall

After that you can install the latest driver and the disable option should work for you. I think :P
If not, at last you uninstalled properly.

Cheers!

BCMNG, 2015-06-23:
Thanks!
I went with the command you've posted earlier:
pkgutil --files "com.mice.pkg.Xbox360controller"

and installed it again. The issue with the game didn't get solved. I'm waiting for their support to answer (Valve/Steam). Went with Bootcamp now. Feels kinda like a loss :/

btw -> restart/uninstall did not work. Showed the same error-button.
Sadly can't provide anything now to help you guys I guess?

masterlui, 2015-06-23:
Good Morning from Colombia,. I'd like to know how safe is to install the driver on my mac using the microsoft wireless receiver for my xbox360 controller? here is an screenshot of my system w/ Yosemite 10.10.3
 
[image]

FranticRain, 2015-06-23:
I have an identical computer, it runs just fine. If you can run Yosemite, you can run this driver.

masterlui, 2015-06-23:
Awesome,. thanks...I'll try again!

RodrigoCard, 2015-07-10:
I will close this and post the beta soon (for real), thanks!

closed 2015-07-10 by RodrigoCard