360Controller/360Controller #99: Ion Drum Rocker - Yosemite 10.10.3 - No devices found

PEMAMETAL, 2015-04-29:
Hello Everyone,

I have been trying to set up a Ion Drum Rocker to use it on Garage Band, I have tried three version of the drivers but none of them seem to work 0.12, 0.14 and 0.15. When I plug the usb cable, the X on the middle of the controller blinks, but when I open the System Preferences it says "No devices found". Am I doing something wrong? 

I am using a MacBook Pro 15"Nvidia 750M, Yosemite 10.10.3

FranticRain, 2015-04-29:
This is a link to my driver which is a separate thing, but we need your vendor ID and product ID for that controller. [These instructions should help you figure out how to get it.](https://github.com/FranticRain/Xone-OSX#troubleshooting)

PEMAMETAL, 2015-04-29:
Hey FranticRain,

Here is what is shows my System Report. I am not sure what I have to do now to set it right
[image]

Drum Rocker:

  Product ID:	0x0130
  Vendor ID:	0x1bad
  Version:	1.00
  Serial Number:	3DE4355
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Ion Audio
  Location ID:	0x14500000 / 9
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

FranticRain, 2015-04-29:
One of the other developers should come along in a while and use this information to compile a version of the driver that works for you. Giving them this information will make the process much faster.

PEMAMETAL, 2015-04-29:
Hey man, thanks for your fast reply. Hope they solve my issue.

PEMAMETAL, 2015-05-04:
So, now I have to download it again and should work?

RodrigoCard, 2015-05-29:
Hello, please check this preview version
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

Cheers!

RodrigoCard, 2015-06-01:
confirmed to be working

closed 2015-06-01 by RodrigoCard