360Controller/360Controller #100: Xbox 360 Pro Ex

mwnickerson, 2015-05-01:
Hi I have an xbox 360 Pro Ex controller (not the mini). I was wondering if you could add it to the list of supported devices. The specs are below. Thanks!

Product id: 0x530a
Vendor ID: 0x24c6
Version: 1.00
serial number: 000124D4
Need: Up to 12mb/sec
Manufacturer: BDA
Location ID: 0xfa13000/7
Current available m(A): 500
Current required m(A): Unkown (device has not been configured)

Thanks again! Let me know if you need anything else!

RodrigoCard, 2015-05-29:
Hello, please check this preview version
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview7.pkg

Cheers!

RodrigoCard, 2015-07-10:
new preview build https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview8.pkg

@mwnickerson could you check this? thanks!

RodrigoCard, 2015-07-10:
well, it should be working, if not, please reopen ;)

closed 2015-07-10 by RodrigoCard