360Controller/360Controller #102: Add option to disable drivers

s1h4d0w, 2015-05-10:
Hi, 

I've recently bought Star Wars: The Force Unleashed and found out that my controller wasn't working, even though the game has explicit Xbox controller support. By Googling I found out that the game has its own drivers and that if you have a different (unofficial) driver it won't work.

So this means I will have to delete the files this driver uses, and when I want to play a different game I will have to reinstall them and repeat the process when I want to play SW:TFU again.

It would be really handy to have a "disable driver" option in the OSX Preferences menu (along with an uninstall option).

Deph0, 2015-05-10:
This feature is already in the works.

FranticRain, 2015-05-10:
There is a beta version of the driver that contains this feature. You can download it here: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview6.pkg

s1h4d0w, 2015-05-10:
Really awesome, thanks!

mele113, 2015-05-20:
Thank You RodigroCard , It works perfectly version 0.15 and YosemiteOSX

RodrigoCard, 2015-07-10:
I will close this and post the beta soon (for real), thanks!

closed 2015-07-10 by RodrigoCard