360Controller/360Controller #105: Install breaks HDI drivers

julian-weinert, 2015-05-23:
Since I installed the latest package, my HDI driver are broken.

After installation I cannot use my internal trackpad and keyboard anymore.

Any ideas how to debug this issue?
In single user mode they keyboard doesn't work either.

Only recovery mode is workung. I tried
`pkgutil -volume /Macintosh\ HD/ -files com.mice.pkg.XBox360Controller` but it says that no receipt was found.

RodrigoCard, 2015-05-23:
````
pkgutil --files "com.mice.pkg.Xbox360controller"
````
the 'c' in controller is not capitalized.
you can find the name of a pkg if you only have a clue by typing:
````
pkgutil --pkgs | grep "something"
````
I would type "360" or any variation of "xbox" "Xbox" "XBox" to find the pkg name, and in fact that was wht I did when I didnt know the exact name :)

------

that said, could you tell us the Vendor/Product IDs of the devices you have that are conflicting?
also, which version or the driver are you using?

julian-weinert, 2015-05-26:
Oh sure... That was written on my mobile from memory, so it probably was a type.
I had to boot from the recovery partition and hat to work on the different drive.

I now reinstalled OS X and it's working again, but sure, I didn't keep the driver.
The computer was broken and I had to fix it quickly so I might cannot make further researches anymore.

My configuration (if you need more, please request):

```plain
USB 3.0 Hi-Speed Bus:

  Host Controller Location:	Built-in USB
  Host Controller Driver:	AppleUSBXHCI
  PCI Device ID:	0x8c31 
  PCI Revision ID:	0x0005 
  PCI Vendor ID:	0x8086 
  Bus Number:	0x0a 

BRCM20702 Hub:

  Product ID:	0x4500
  Vendor ID:	0x0a5c  (Broadcom Corp.)
  Version:	1.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14800000 / 3
  Current Available (mA):	500
  Current Required (mA):	94
  Built-In:	Yes

Bluetooth USB Host Controller:

  Product ID:	0x8289
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	1.03
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14830000 / 6
  Current Available (mA):	500
  Current Required (mA):	0
  Built-In:	Yes

Apple Internal Keyboard / Trackpad:

  Product ID:	0x0263
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	2.25
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14c00000 / 2
  Current Available (mA):	500
  Current Required (mA):	40
  Built-In:	Yes

BRCM20702 Hub:

  Product ID:	0x4500
  Vendor ID:	0x0a5c  (Broadcom Corp.)
  Version:	1.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14800000 / 3
  Current Available (mA):	500
  Current Required (mA):	94
  Built-In:	Yes

Bluetooth USB Host Controller:

  Product ID:	0x8289
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	1.03
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14830000 / 6
  Current Available (mA):	500
  Current Required (mA):	0
  Built-In:	Yes

Bluetooth USB Host Controller:

  Product ID:	0x8289
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	1.03
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14830000 / 6
  Current Available (mA):	500
  Current Required (mA):	0
  Built-In:	Yes

Apple Internal Keyboard / Trackpad:

  Product ID:	0x0263
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	2.25
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14c00000 / 2
  Current Available (mA):	500
  Current Required (mA):	40
  Built-In:	Yes

USB 3.0 SuperSpeed Bus:

  Host Controller Location:	Built-in USB
  Host Controller Driver:	AppleUSBXHCI
  PCI Device ID:	0x8c31 
  PCI Revision ID:	0x0005 
  PCI Vendor ID:	0x8086 
  Bus Number:	0x0a 

Internal Memory Card Reader:

  Product ID:	0x8406
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	8.20
  Speed:	Up to 5 Gb/sec
  Manufacturer:	Apple
  Location ID:	0x15400000 / 1
  Current Available (mA):	900
  Current Required (mA):	896
  Capacity:	63.86 GB (63,864,569,856 bytes)
  Removable Media:	Yes
  Detachable Drive:	Yes
  BSD Name:	disk2
  Built-In:	Yes
  Partition Map Type:	MBR (Master Boot Record)
  S.M.A.R.T. status:	Not Supported
  Volumes:
Macintosh SD:
  Capacity:	63.85 GB (63,847,792,640 bytes)
  Available:	57.56 GB (57,558,695,936 bytes)
  Writable:	Yes
  File System:	Journaled HFS+
  BSD Name:	disk2s1
  Mount Point:	/Volumes/Macintosh SD
  Content:	Apple_HFS

Internal Memory Card Reader:

  Product ID:	0x8406
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	8.20
  Serial Number:	000000000820
  Speed:	Up to 5 Gb/sec
  Manufacturer:	Apple
  Location ID:	0x15400000 / 1
  Current Available (mA):	900
  Current Required (mA):	896
  Capacity:	63.86 GB (63,864,569,856 bytes)
  Removable Media:	Yes
  Detachable Drive:	Yes
  BSD Name:	disk2
  Built-In:	Yes
  Partition Map Type:	MBR (Master Boot Record)
  S.M.A.R.T. status:	Not Supported
  Volumes:
Macintosh SD:
  Capacity:	63.85 GB (63,847,792,640 bytes)
  Available:	57.56 GB (57,558,695,936 bytes)
  Writable:	Yes
  File System:	Journaled HFS+
  BSD Name:	disk2s1
  Mount Point:	/Volumes/Macintosh SD
  Content:	Apple_HFS
```

RodrigoCard, 2015-07-10:
Sorry for not leaving feedback before but, Which are your HDI driver IDs? I cant recognize from that list alone.

Also, You said you used the 0.14 version, please check this one:
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview8.pkg

if you cant get it to work, this version includes a uninstaller, just uninstall and reboot.

cheers

RodrigoCard, 2016-01-19:
no response, closing.

closed 2016-01-19 by RodrigoCard