360Controller/360Controller #106: Disconecting the xbox controller from Mac

rrrvxi, 2015-05-31:
Hello, I installed the most recent version 0.15, it works with Yosemite 10.10.3, I have the official MS wireless for PC. All good, but does anybody knows if there a way to disconnect the controller without having to go to system preferences and then push the power off? I ask because in Windows I had a 3rd party program that just clicking on it automatically disconnected the control, in fact I remember I made a keyboard shortcut so it was even more efficient and FAST. If anyone know if this is possible on Mac please advise. Thanks

RodrigoCard, 2015-05-31:
A Menu Bar for that and other functions would be great.
but for now, I suggest ou just pop out the batteries box, it is faster :)

RodrigoCard, 2015-07-28:
This is duplicate of #47

closed 2015-07-28 by RodrigoCard