360Controller/360Controller #107: QanBa Joystick Plus support (Q1 for 360)

Geogi, 2015-06-03:
Hi,

My stick isn't detected, is it possible to add support for it?
(NB: it's the 360 version of that stick, not the PS version; it works on Linux with `xboxdrv`)

Here's the system info.
QanBa Joystick Plus:

  Product ID:	0xbeef
  Vendor ID:	0x0738  (Mad Catz, Inc.)
  Version:	1.14
  Serial Number:	069497B1
  Speed:	Up to 12 Mb/sec
  Manufacturer:	HJC
  Location ID:	0x14500000 / 8
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

RodrigoCard, 2015-07-10:
Please download and install this one and tell us if is worked
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview8.pkg

I hope to release a proper beta soon (this time for real :P )

RodrigoCard, 2015-07-10:
well, it should be working, if not, please reopen ;)

closed 2015-07-10 by RodrigoCard