360Controller/360Controller #109: Change repo description to give it more visibility for it in Github search

giovannibonetti, 2015-06-15:
Hi! Congratulations for the project!
When I tried to find it on Github, I searched for "xbox 360 mac" and found nothing. Then I tried "360 controller mac" and still nothing. Only when I tried "360 controller" it showed up.
Perhaps you could add the word "Mac" in the description. Apparently the readme doesn't count for Github search.

Just a suggestion.

RodrigoCard, 2015-06-17:
Agreed, I would change it to:
```
Xbox 360, Xbox Original and Xbox One Game Controller Driver for Mac OS X (based on TattieBogle's work with improvements) 
```

Or something as "Search Engine Optimized" as someone elase can think...
Anyway, I cant change that.
@d235j  can.

RodrigoCard, 2015-07-10:
Ignore the last comment, that was ugly.
@d235j Could you change the repo description to something like this?

```
Xbox 360 / Xbox Original / Xbox One Game Controller Driver for Mac OS X (forked from TattieBogle's work)
```

Thank you! :)

slawo, 2015-10-08:
You should also add a link to this project in your own Test repo and in TattieBogle's repo.