360Controller/360Controller #110: Mad Catz Pro Controller (similar to the MLG)

tijames, 2015-06-19:
Can this one be added? Having some trouble with it on yosemite.

MLG GamePad for Xbox 360:

  Product ID:	0xf023
  Vendor ID:	0x1bad
  Version:	4.90
  Serial Number:	19197C16
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Mad Catz, Inc.
  Location ID:	0x14800000 / 20
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

RodrigoCard, 2015-06-19:
we will add soon

tijames, 2015-07-04:
Any idea when their will be a release with the controller added to the list or maybe a beta version? I'm eager to try it out. :D

RodrigoCard, 2015-07-10:
"SOON" was not that close, haha, I finally got some time to this project, I will add that.

RodrigoCard, 2015-07-10:
Please download and install this one and tell us if is worked
https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview8.pkg

I hope to release a proper beta soon (this time for real :P )

RodrigoCard, 2015-07-10:
well, it should be working, if not, please reopen ;)

closed 2015-07-10 by RodrigoCard

tijames, 2015-07-10:
Works perfectly. Thanks :D