360Controller/360Controller #112: "No devices found" on Yosemite 10.10.3

shadowycore, 2015-06-25:
I have installed the v0.14-unofficial release and my wireless receiver doesn't seem to be working. Under the "About This Mac >> System Report >> Hardware >> USB" tab there is nothing indicating a USB controller. The light on the receiver does not turn on, and the controller refuses to pair with it. I'm not sure what I'm doing wrong here.

The pane in System Preferences opens just fine, but does not list any controllers.

Any help would be greatly appreciated!

d235j, 2015-06-25:
If the wireless receiver is not appearing at all in System Profiler, it's probably defective. It should appear in some form whether or not the driver is installed.

shadowycore, 2015-06-26:
Well I brought it to work where I had a PC to try it on and it looks like you're right. Not sure how this happened, but I guess I'll be replacing it. Thanks.

closed 2015-06-26 by shadowycore

shadowycore, 2015-06-26:
I just wanted to confirm that this was indeed the issue. I replaced the receiver and all is working now.