360Controller/360Controller #113: Controller not recognised as a gamepad in Elite

WaferMouse, 2015-06-29:
Elite: Dangerous provides special control schemes for gamepads, but it doesn't detect this driver as a gamepad.  For an example of one that does, see the XBox One controller driver at https://github.com/FranticRain/Xone-OSX

Confirmed in 0.15Preview6 (which is a massive improvement over previous versions, thanks!)

FranticRain, 2015-06-29:
That is extremely ironic, since this driver is recognized by WAY more games than mine is. It seems crazy to put out a controller enabled game and not directly support this driver.

The main differences between the two is that the Xone-OSX driver is an IOUSBHIDDevice, while this one is an IOHIDDevice. It seems almost as if this is a game-level issue, not a driver-level one. I don't know for sure though.

slawo, 2015-10-08:
That driver (https://github.com/FranticRain/Xone-OSX) is deprecated. The game will have to be updated to support the replacement driver.

This issue should probably be fixed in the game. Maybe someone from [Frontier](http://www.frontier.co.uk) could comment on it if the issue was driver related.

electric-monk, 2015-10-08:
There are some games which specifically look for the class names of this driver. I discovered this after doing some code feng shui and breaking them, which required even more intense code feng shui to put it back. It totally wouldn't surprise me if Frontier were doing something odd.

Looking at the game's imported symbols it looks like it's just using plain HID methods, but I've not yet figured out what specifically it might be doing to cause such an issue. I'll have a more detailed look if I find some time.

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain