360Controller/360Controller #115: Write the content of the About tab

RodrigoCard, 2015-07-10:
Could someone help me to write proper credits in place of that Lorem Ipsum in the About tab?
On top of my head, some names:
@d235j @MaddTheSane @FranticRain 
and Tattiebogle of course

should we put real names or just github nicks?
maybe contacts/site?
some lovely text? :P

Please, ideas! Thanks!

RodrigoCard, 2015-07-10:
Text pasted from the Welcome text of the installer I just updated, it may help:
Update 0.15 (beta): XX/XX/2015
This update includes:
	•	Completely reworked User Interface by Pyroh
	•	Xbox One Controller Support with Force Feedback by FranticRain.
	•	Added dozens of device IDs provided by users, so most of Xbox compatible controllers should work
	•	Swappable button ids and Deadzone adjustment improvements by FranticRain
	•	General bug fixes and improvements mostrly from C.W. Betts
	•	You can now temporarily disable or completely uninstall the driver from the Preference Panel itself by RodrigoCard.

Improvements to this text are welcome too.

RodrigoCard, 2015-07-25:
So, what you think? https://github.com/d235j/360Controller/commit/2c5dc5d5526e308d589ed6a36fdac1f919b4756c

[image]

If there is nothing wrong I will release a beta and put the release page as is :)

Also, Tattlebogle is back, aparently, https://github.com/electric-monk/360Controller/
I will talk to him :)

d235j, 2015-07-25:
> Also, Tattlebogle is back, aparently, https://github.com/electric-monk/360Controller/
> I will talk to him :)

Oh wow, finally.
We should pull-request this work to his repo, as long as he's willing to maintain it. :)

RodrigoCard, 2015-07-26:
I totally agree! I will just check #101 and build a beta version tomorrow to release and then ask him. :)

RodrigoCard, 2015-07-28:
https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial
Now I will sleep a bit before writing to him... -.-

closed 2015-07-28 by RodrigoCard