360Controller/360Controller #116: Xbox One Controllers with 3.5mm Jack Don't Work. (Covert Forces)

FranticRain, 2015-07-11:
Despite the added product and vendor IDs, the new controllers aren't responding to the init packets. I had one user sniff the packets, and made a driver that sent those exact packets back. That implementation did not wake the controller. Unfortunately, it appears that I will have to buy one of the new controllers. I really hope I don't have to buy an Elite controller too.

[For more information, you can check out the the issue on my repo.](https://github.com/FranticRain/Xone-OSX/issues/63)

RodrigoCard, 2015-07-28:
since it is not working I removed the ID from the plist temporarily, so I can release a 0.15 beta.
We should add that again once it is working :)

FranticRain, 2015-07-28:
Sounds good to me. I really just need to buy one of the new controllers, but its hard to justify it when I have two old ones, and no reliable source of income. Yay, full time student!

RodrigoCard, 2015-08-28:
fixed by #120

closed 2015-08-28 by RodrigoCard

RodrigoCard, 2015-08-28:
https://github.com/d235j/360Controller/releases/tag/v0.15_beta2_unofficial